mightymos/OnbrightFlasher #7: Can't compile and flash to ESP32

paulandrewcrouch, 2024-11-19:
I've tried on VS Code and Ardunio IDE to compile the flasher.   I've set the pins to 21 and 22, and I keep getting this as a warning

```
OnbrightFlasher.ino: In function 'uint8_t state_machine_command(int, uint8_t)':
OnbrightFlasher.ino:480:61: error: no matching function for call to 'OnbrightFlasher::readFlashBlock(int, uint8_t [32767], int)'
       flasher.readFlashBlock(0, fileArray, TARGET_FLASH_SIZE);
```
Any help you can give would be appreciated.

mightymos, 2024-11-20:
That's really strange, did you clone the project with git or download the entire zip?

That function is located in the onbrightFlasher.cpp source file.

I am using Arduino IDE version 2.3.3 and do not receive a compilation error.

mightymos, 2024-12-02:
Closing for now as cannot reproduce.
If any additional information is available please advise.

closed 2024-12-02 by mightymos