mightymos/OnbrightFlasher #10: Compilation error: parameter 'configByte' includes reference to array of unknown bound 'unsigned char []'

X-GeneIT, 2024-12-01:
I'm unable to compile the code in Arduino IDE 2.3.4/2.3.3. 
I downloaded the zip file from git then extracted it to OnbrightFlasher folder.
I'm using kuongshun uno r3
SDA 16
SCL 17
Any help would be highly appreciated! 

`In file included from D:\Dev\Microcontrollers\OnbrightFlasher\onbrightFlasher.cpp:7:0:
D:\Dev\Microcontrollers\OnbrightFlasher\onbrightFlasher.h:91:112: error: parameter 'configByte' includes reference to array of unknown bound 'unsigned char []'
     byte readConfigBlock(const unsigned char address, unsigned char (&configByte)[], const unsigned char length);
                                                                                                                ^
D:\Dev\Microcontrollers\OnbrightFlasher\onbrightFlasher.h:96:113: error: parameter 'flashbyte' includes reference to array of unknown bound 'unsigned char []'
     byte readFlashBlock(const unsigned int flashAddress, unsigned char (&flashbyte)[], const unsigned int length);
                                                                                                                 ^
D:\Dev\Microcontrollers\OnbrightFlasher\onbrightFlasher.cpp:235:126: error: parameter 'flashbyte' includes reference to array of unknown bound 'unsigned char []'
 byte OnbrightFlasher::readFlashBlock(const unsigned int flashAddress, unsigned char (&flashbyte)[], const unsigned int length)
                                                                                                                              ^
D:\Dev\Microcontrollers\OnbrightFlasher\onbrightFlasher.cpp:273:129: error: parameter 'flashbyte' includes reference to array of unknown bound 'unsigned char []'
 byte OnbrightFlasher::readConfigBlock(const unsigned char flashAddress, unsigned char (&flashbyte)[], const unsigned char length)
                                                                                                                                 ^
In file included from D:\Dev\Microcontrollers\OnbrightFlasher\OnbrightFlasher.ino:8:0:
D:\Dev\Microcontrollers\OnbrightFlasher\onbrightFlasher.h:91:112: error: parameter 'configByte' includes reference to array of unknown bound 'unsigned char []'
     byte readConfigBlock(const unsigned char address, unsigned char (&configByte)[], const unsigned char length);
                                                                                                                ^
D:\Dev\Microcontrollers\OnbrightFlasher\onbrightFlasher.h:96:113: error: parameter 'flashbyte' includes reference to array of unknown bound 'unsigned char []'
     byte readFlashBlock(const unsigned int flashAddress, unsigned char (&flashbyte)[], const unsigned int length);
                                                                                                                 ^
D:\Dev\Microcontrollers\OnbrightFlasher\OnbrightFlasher.ino: In function 'uint8_t state_machine_command(int, uint8_t)':
D:\Dev\Microcontrollers\OnbrightFlasher\OnbrightFlasher.ino:482:61: error: no matching function for call to 'OnbrightFlasher::readFlashBlock(int, uint8_t [32767], int)'
       flasher.readFlashBlock(0, fileArray, TARGET_FLASH_SIZE);
                                                             ^
In file included from D:\Dev\Microcontrollers\OnbrightFlasher\OnbrightFlasher.ino:8:0:
D:\Dev\Microcontrollers\OnbrightFlasher\onbrightFlasher.h:96:10: note: candidate: byte OnbrightFlasher::readFlashBlock(unsigned int, unsigned char (&)[], unsigned int)
     byte readFlashBlock(const unsigned int flashAddress, unsigned char (&flashbyte)[], const unsigned int length);
          ^~~~~~~~~~~~~~
D:\Dev\Microcontrollers\OnbrightFlasher\onbrightFlasher.h:96:10: note:   no known conversion for argument 2 from 'uint8_t [32767] {aka unsigned char [32767]}' to 'unsigned char (&)[]'
D:\Dev\Microcontrollers\OnbrightFlasher\OnbrightFlasher.ino:497:63: error: no matching function for call to 'OnbrightFlasher::readConfigBlock(int, uint8_t [255], int)'
       flasher.readConfigBlock(0, configBytes, CONFIG_BYTE_SIZE);
                                                               ^
In file included from D:\Dev\Microcontrollers\OnbrightFlasher\OnbrightFlasher.ino:8:0:
D:\Dev\Microcontrollers\OnbrightFlasher\onbrightFlasher.h:91:10: note: candidate: byte OnbrightFlasher::readConfigBlock(unsigned char, unsigned char (&)[], unsigned char)
     byte readConfigBlock(const unsigned char address, unsigned char (&configByte)[], const unsigned char length);
          ^~~~~~~~~~~~~~~
D:\Dev\Microcontrollers\OnbrightFlasher\onbrightFlasher.h:91:10: note:   no known conversion for argument 2 from 'uint8_t [255] {aka unsigned char [255]}' to 'unsigned char (&)[]'
D:\Dev\Microcontrollers\OnbrightFlasher\OnbrightFlasher.ino: In function 'void setup()':
D:\Dev\Microcontrollers\OnbrightFlasher\OnbrightFlasher.ino:624:28: error: no matching function for call to 'TwoWire::begin(int&, int&)'
   Wire.begin(sdaPin, sclPin);
                            ^
In file included from D:\Dev\Microcontrollers\OnbrightFlasher\OnbrightFlasher.ino:33:0:
C:\Users\\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.8.6\libraries\Wire\src/Wire.h:53:10: note: candidate: void TwoWire::begin()
     void begin();
          ^~~~~
C:\Users\\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.8.6\libraries\Wire\src/Wire.h:53:10: note:   candidate expects 0 arguments, 2 provided
C:\Users\\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.8.6\libraries\Wire\src/Wire.h:54:10: note: candidate: void TwoWire::begin(uint8_t)
     void begin(uint8_t);
          ^~~~~
C:\Users\\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.8.6\libraries\Wire\src/Wire.h:54:10: note:   candidate expects 1 argument, 2 provided
C:\Users\\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.8.6\libraries\Wire\src/Wire.h:55:10: note: candidate: void TwoWire::begin(int)
     void begin(int);
          ^~~~~
C:\Users\\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.8.6\libraries\Wire\src/Wire.h:55:10: note:   candidate expects 1 argument, 2 provided

exit status 1

Compilation error: parameter 'configByte' includes reference to array of unknown bound 'unsigned char []'`

X-GeneIT, 2024-12-01:
PIN numbers updated

X-GeneIT, 2024-12-01:
I changed the parameter to a pointer instead of a reference array. then it exceeded available space in the board.

mightymos, 2024-12-02:
> I changed the parameter to a pointer instead of a reference array. then it exceeded available space in the board.

To read out the entire flash space of the OB38S003 into ram requires 8KB but the Uno R3 only has 2KB.
Admittedly all we are doing with the contents now is computing the checksum, though someone might want to obtain the full contents in the future to fully verify flashing was successful.

For now, if you just comment out any lines related to fileArray[] I think you can fit within the space of the Uno R3.

Let me know if it works?

X-GeneIT, 2024-12-02:
Thank you for your time and support. much appreciated! I managed to flash it using ESP8866 board and your custom firmware (Tasmota + Portisch). I receive codes from my garage roller shutter remote under raw sniff mode. however, I'm struggling to send code from the device. am I missing any mapping? 

`12:48:26.886 MQT: tele/tasmota_rf_bridge_gf/RESULT = {"Time":"2024-12-02T12:48:26","RfRaw":{"Data":"AA B1 04 0171 0EFB 02EF 3DD5 3808080808080808080808081A082A0828282A0A0A0A082A0A082A082A082A082A082A08282A082828282A082A0A082A0828282A0A0828282A0A08282828282A0A082A0A0A082A0A0A082A0A0828 55"}}
12:48:27.200 MQT: tele/tasmota_rf_bridge_gf/RESULT = {"Time":"2024-12-02T12:48:27","RfRaw":{"Data":"AA B1 04 016F 0EFB 02EE 3DD5 3808080808080808080808081A082A0828282A0A0A0A082A0A082A082A082A082A082A08282A082828282A082A0A082A0828282A0A0828282A0A08282828282A0A082A0A0A082A0A0A082A0A0828 55"}}
12:48:27.512 MQT: tele/tasmota_rf_bridge_gf/RESULT = {"Time":"2024-12-02T12:48:27","RfRaw":{"Data":"AA B1 04 016E 0EFD 02EF 3DD7 3808080808080808080808081A082A0828282A0A0A0A082A0A082A082A082A082A082A08282A082828282A082A0A082A0828282A0A0828282A0A08282828282A0A082A0A0A082A0A0A082A0A0828 55"}}
12:48:27.827 MQT: tele/tasmota_rf_bridge_gf/RESULT = {"Time":"2024-12-02T12:48:27","RfRaw":{"Data":"AA B1 04 016F 0EFB 02EF 3DD5 3808080808080808080808081A082A0828282A0A0A0A082A0A082A082A082A082A082A08282A082828282A082A0A082A0828282A0A0828282A0A08282828282A0A082A0A0A082A0A0A082A0A0828 55"}}
12:48:28.143 MQT: tele/tasmota_rf_bridge_gf/RESULT = {"Time":"2024-12-02T12:48:28","RfRaw":{"Data":"AA B1 04 016E 0EFE 02EF 3DD4 3808080808080808080808081A082A0828282A0A0A0A082A0A082A082A082A082A082A08282A082828282A082A0A082A0828282A0A0828282A0A08282828282A0A082A0A0A082A0A0A082A0A0828 55"}}`

`{"NAME":"Generic","GPIO":[32,3200,1,3232,1,1,0,0,1,320,1,0,0,0],"FLAG":0,"BASE":25}`

mightymos, 2024-12-02:
Is that the code you want to replay?

You need to put them through a tool to get the corresponding 0xB0 command for transmitting like explained here:
https://tasmota.github.io/docs/devices/Sonoff-RF-Bridge-433/#b1-to-b0-conversion-tools

Then transmit using rfraw command.

mightymos, 2024-12-02:
Cool, glad it is working through  Tasmota console.
I don't currently know much of anything about MQTT.

You might be able to open an issue over on the firmware project and possibly someone would see it there that could help.
But you might have to reference that issue over in the Home Assistant or Mosquitto projects to get help.

If you find out more you can let me know so I can learn also.
But closing this issue now as flasher was able to work.

closed 2024-12-02 by mightymos