mightymos/OnbrightFlasher #11: Flashing RF-Chip doesn't work

JMDesigns123, 2025-01-10:
Hello!

I need help to flash the HEX-File with the Onbrightflasher to the RF-Chip. The step before with erasing the RF-Chip was successful, but when i start flashscript.py, after the successful handshake comes the following error:

2025-01-10 16:18:40,975 - INFO - You selected: C:\Users\juerg\downloads\onbrightflasher\main_portisch_OB38S003.hex
2025-01-10 16:18:42,086 - INFO - handshake
2025-01-10 16:18:42,088 - INFO - State changing to handshake
2025-01-10 16:18:42,091 - INFO - cycle power to target (start with power off and then turn on)
2025-01-10 16:18:49,843 - INFO - Handshake succeeded
2025-01-10 16:18:49,963 - INFO - Status: 2
2025-01-10 16:18:49,964 - ERROR - Something went wrong. Process ended with Status: 2
2025-01-10 16:18:49,964 - ERROR - State 4 failed. Returning to handshake state.
2025-01-10 16:18:49,966 - INFO - NACK on transmit of address
2025-01-10 16:18:49,967 - INFO - Chip read type FAILED
2025-01-10 16:18:49,968 - INFO - Chip type reported was: 0xD0
2025-01-10 16:18:49,975 - INFO - Can try command [signature] or [idle] then [handshake] to retry
2025-01-10 16:18:49,975 - ERROR - Something went wrong. Process ended with Can try command [signature] or [idle] then [handshake] to retry

I hope some can help me to solve the problem. Thank you very much in advance!

lg
Jürgen

mightymos, 2025-01-10:
Did you erase the microcontroller manually?

If so, you do not need to do it manually.
Just let the script perform the entire procedure.

You just need to apply power to the microcontroller separately when it loops "cycle power ..." in the script.

JMDesigns123, 2025-01-10:
I erased the microcontroller manually, how it is in the flashing guide described. 

How is the cabling between Wemos D1 and the microcontroller? I used only SDA und SCL and the power supply for the microcontroller was with USB-Cable. This worked when i erased the microcontroller.

mightymos, 2025-01-10:
Ok, that seems correct, especially if it worked previously with manual erase.
Let me try the script later today to confirm if it remains working or not.

Janrupf, 2025-01-11:
I just tried too, and I'm getting the same error (interestingly enough also Chip id `0xD0`). Did they possibly change something again?

mightymos, 2025-01-11:
> I just tried too, and I'm getting the same error (interestingly enough also Chip id `0xD0`). Did they possibly change something again?
> 
> Side note: I'm not using (well, didn't try yet) the python script, so I haven't yet erased the RF chip, only the ESP on the board.

I might take me longer than I thought to look into this.
But it'd be good to check, what is written on the silkscreen on the board?

Mine says:
```
433RFBRIGE R2 V2.2
2021.01.19
```

And if you can read the markings on the microcontroller with a magnifying glass too that would also help.

Janrupf, 2025-01-11:
Layer 8 problem... try checking the SCL line. If you got the wrong pin (I used SCLK instead of SCL!) the read still partially works, but I guess the clock doesn't fully sync up and weird stuff happens. So at least in my case it was just the clock line not being correct. (Note: on the NodeMCU its D1 for SCL, D2 for SDA, they are next to each other)

JMDesigns123, 2025-01-12:
I have the same imprint:

433RFBRIDGE R2 V2.2
2021.01.19

Which one is the microcontroller. Is this the chip over the eight pins? Opposite of the pins USBTXD and RXD?

JMDesigns123, 2025-01-12:
Thanks for the tip, I take the following board to flash the microcontroller:

[image]

It doesn't work with SCL and it doesn't word with SCK. Do you thinkg wihich pins i should take to flash the microcontroller?

mightymos, 2025-01-12:
> Thanks for the tip, I take the following board to flash the microcontroller:
> 
> 
> It doesn't work with SCL and it doesn't work with SCK. Do you thinkg wihich pins i should take to flash the microcontroller?

SCK is the wrong pin to use from that board, you need to use SDA and SCL.
I do not understand why that board shows GPIO4 (SDA) and GPIO5 (SCL) twice.

JMDesigns123, 2025-01-12:
Which board do you use to flash the microcontroller?

mightymos, 2025-01-12:
> I used SDA und SCL at the beginnend to flash. It doesn't work. Which board do you use to flash the microcontroller?

I used the Wemos D1 mini but I think it was the v3:
https://www.wemos.cc/en/latest/d1/d1_mini.html

I noticed that the pins SCL and SDA are swapped in the diagram you posted.
So D15 is SCL which is top most.
But on the lower grouping of pins D4 is SDA.

Can you confirm which pins you actually hooked up to the microcontroller header?

JMDesigns123, 2025-01-12:
I used the pins D14 (SDA) und D15 (SCL) and it doesn't work with this pins. The step before with erasing the chip with arduino, i also used this two pins and it did work to erase the chip. 

Now i used the pins D3 (SCL) and D4 (SDA) and it's also didnt't work.

mightymos, 2025-01-12:
> I used the pins D14 (SDA) und D15 (SCL) and it doesn't work with this pins. The step before with erasing the chip with arduino, i also used this two pins and it did work to erase the chip.
> 
> Now i used the pins D3 (SCL) and D4 (SDA) and it's also didnt't work.

Are you powering the Wemos D1 with the barrel jack or the USB port?

JMDesigns123, 2025-01-12:
With the USB port. For a successful handshake i powering the microcontroller also with the USB port when the command "cycling power..." is coming.

mightymos, 2025-01-12:
I believe this is similar to your board and shows a schematic:
https://wiki.geekworm.com/WEMOS_ESP8266_D1_R2_V2.1

I think you need to connect the GND, SCL, SDA from the Wemos D1 to the same labeled pins on Sonoff.
Then run the script.
Once the script asks you to supply power, connect 3.3V from the Wemos D1 to your Sonoff.
The flashing should then continue.
If it does not disconnect 3.3V and reconnect it a few times.

The Wemos D1 apparently has an RT9013 3.3V regulator that can supply 500 mA.
That should be plenty for powering the Sonoff during flashing:
https://www.richtek.com/Products/Linear%20Regulator/Single%20Output%20Linear%20Regulator/RT9013?sc_lang=en&specid=RT9013

JMDesigns123, 2025-01-12:
Thank's it works now!!!

mightymos, 2025-01-12:
> Thank's it works now!!!

Fantastic. You can also post on the other project once you try to decode your radio devices.

closed 2025-01-14 by mightymos