mightymos/OnbrightFlasher #12: Hardware mod question

ipsumium, 2025-02-02:
Hi, thanks for the in detail article!
Do I still have to perform the hardware mod?
Until now i have successfully flashed the file: main_portisch_OB38S003_MULTI_PROTOCOLS_INCLUDED.hex
I have configured tasmota in Sonoff bridge (25) mode and when i use the rfraw 177 comand i get no signals:

16:45:24.090 RSL: RESULT = {"RfRaw":"ON"}
16:45:24.119 RSL: RESULT = {"Time":"2025-02-02T16:45:24","RfRaw":{"Data":"AAA055"}}
16:45:42.984 CMD: RfRaw 177
16:45:42.989 RSL: RESULT = {"RfRaw":"ON"}
16:45:43.010 RSL: RESULT = {"Time":"2025-02-02T16:45:43","RfRaw":{"Data":"AAA055"}}
16:45:48.290 CMD: RfRaw 177
16:45:48.295 RSL: RESULT = {"RfRaw":"ON"}
16:45:48.323 RSL: RESULT = {"Time":"2025-02-02T16:45:48","RfRaw":{"Data":"AAA055"}}
16:45:59.610 CMD: RfRaw 177
16:45:59.615 RSL: RESULT = {"RfRaw":"ON"}
16:45:59.643 RSL: RESULT = {"Time":"2025-02-02T16:45:59","RfRaw":{"Data":"AAA055"}}

Thanks for your time!

mightymos, 2025-02-02:
rfraw 177 is equivalent to rfraw AA B1 55 which is the sniffing command (i.e. 0xB1).

In order to sniff codes with unknown format/timing you need to flash the firmware that has sniffing included in the filename.

ipsumium, 2025-02-11:
Thanks a lot, works as expected after flashing with sniffing included fw

mightymos, 2025-02-12:
Just out of curiosity, what output do you see and from what type of devices?

ipsumium, 2025-02-13:
I'm currently using it to control the motor of my window blinds with mqtt messages.
The codes that i'm getting are like this:
07:31:46.501 MQT: stat/tasmota_1B2976/RESULT = {"RfRaw":"ON"}
07:31:46.556 MQT: tele/tasmota_1B2976/RESULT = {"Time":"2025-02-13T07:31:46","RfRaw":{"Data":"AAA055"}}
07:31:50.776 MQT: tele/tasmota_1B2976/RESULT = {"Time":"2025-02-13T07:31:50","RfRaw":{"Data":"AA B1 05 12BC 05BE 0165 02BC 1E51 481A3B2B2A3A3B2A3A3B2B2B2B2B2A3A3B2A3A3A3B2B2A3B2A3A3A3A3A3A3B2A3B2A3A3B2B2A3A3B2B 55"}}
07:31:51.063 MQT: tele/tasmota_1B2976/RESULT = {"Time":"2025-02-13T07:31:51","RfRaw":{"Data":"AA B1 05 12BF 05BE 0167 02BA 1FC0 481A3B2B2A3A3B2A3A3B2B2B2B2B2A3A3B2A3A3A3B2B2A3B2A3A3A3A3A3A3B2A3B2A3A3B2B2B2B2A3A 55"}}
07:31:51.743 MQT: tele/tasmota_1B2976/RESULT = {"Time":"2025-02-13T07:31:51","RfRaw":{"Data":"AA B1 05 12B9 05BB 0166 02BA 1E4C 481A3B2B2A3A3B2A3A3B2B2B2B2B2A3A3B2A3A3A3B2B2A3B2A3A3A3A3A3A3B2A3B2A3B2A3B2A3B2A3B 55"}}

I had to convert them to B0 to replay them.
Remote photos:
https://ibb.co/JWf9Z88F
https://ibb.co/LdTVMkbC

mightymos, 2025-02-14:
HIP01W remote, okay.
Maybe I'll try to look it up, but may not be necessary since you got the codes.

note: The uploaded ibb have GPS data included so you can delete them.

closed 2025-02-14 by mightymos