mightymos/OnbrightFlasher #13: Chip read type FAILED when trying to flash the radio

pszober, 2025-03-18:
Hi,

I followed the instructions, using standard Wemos D1 board flashed from Arduino as per instructions, wired to the RFbridge and connected with 3 wires SCL, SDA and GND.

The RDbridge board is then powered from externam micro usb supply (initially disconnected). Then I start the script, select port and firmware file, connect USB powerr supply to the RF bridge and get this:

```
PS C:\Users\pszob\Downloads\OnbrightFlasher> python flashScript.py
Available COM ports:
1. COM1 - Communications Port (COM1)
2. COM5 - USB-SERIAL CH340 (COM5)
Select a COM port (enter the number): 2
2025-03-18 18:02:36,838 - INFO - Serial connection opened on COM5
Select a file by number:
1. blink.ihx
2. portisch_main_OB38S003_MULTIPLE_PROTOCOLS_INCLUDED.hex
Enter the number of the file you want to select: 2
2025-03-18 18:02:37,973 - INFO - You selected: C:\Users\pszob\Downloads\OnbrightFlasher\portisch_main_OB38S003_MULTIPLE_PROTOCOLS_INCLUDED.hex
2025-03-18 18:02:39,091 - INFO - handshake
2025-03-18 18:02:39,091 - INFO - State changing to handshake
2025-03-18 18:02:39,093 - INFO - cycle power to target (start with power off and then turn on)
2025-03-18 18:02:42,795 - INFO - Handshake succeeded
2025-03-18 18:02:42,915 - INFO - Status: 4
2025-03-18 18:02:42,916 - INFO - Other error
2025-03-18 18:02:42,918 - INFO - Chip read type FAILED
2025-03-18 18:02:42,920 - INFO - Chip type reported was: 0xD0
2025-03-18 18:02:42,926 - INFO - Can try command [signature] or [idle] then [handshake] to retry
2025-03-18 18:02:42,926 - ERROR - Something went wrong. Process ended with Can try command [signature] or [idle] then [handshake] to retry
2025-03-18 18:02:42,926 - ERROR - State 4 failed. Returning to handshake state.
2025-03-18 18:02:44,042 - INFO - handshake
2025-03-18 18:02:44,043 - INFO - State changing to handshake
2025-03-18 18:02:44,044 - INFO - cycle power to target (start with power off and then turn on)
2025-03-18 18:02:44,853 - INFO - Handshake succeeded
2025-03-18 18:02:44,973 - INFO - Status: 3
2025-03-18 18:02:44,975 - INFO - NACK on transmit of data
2025-03-18 18:02:44,976 - INFO - Chip read type FAILED
2025-03-18 18:02:44,978 - INFO - Chip type reported was: 0xD0
2025-03-18 18:02:44,985 - INFO - Can try command [signature] or [idle] then [handshake] to retry
2025-03-18 18:02:44,992 - ERROR - Something went wrong. Process ended with Can try command [signature] or [idle] then [handshake] to retry
2025-03-18 18:02:44,992 - ERROR - State 4 failed. Returning to handshake state.
2025-03-18 18:02:46,099 - INFO - handshake
2025-03-18 18:02:46,100 - INFO - State changing to handshake
2025-03-18 18:02:46,101 - INFO - cycle power to target (start with power off and then turn on)
Serial connection closed.
```

Any ideas how to troubleshoot?

mightymos, 2025-03-18:
Ok, it could matter what particular Wemos D1 board you have.
At least we can double check.

Do you have a link to the particular Wemos D1 you are using?

mightymos, 2025-03-18:
Ok there are several boards linked there, is it with CH9102X, CP2102, CH340 or something else?
I am just trying to confirm the pinout.

What is the 3.3V voltage regulator onboard?
If it has enough current output capacity you can probably power the Sonoff at 3.3V with that.

pszober, 2025-03-18:
Right so the output from Arduino is this:

```
. Variables and constants in RAM (global, static), used 62888 / 80192 bytes (78%)
║   SEGMENT  BYTES    DESCRIPTION
╠══ DATA     1520     initialized variables
╠══ RODATA   1960     constants       
╚══ BSS      59408    zeroed variables
. Instruction RAM (IRAM_ATTR, ICACHE_RAM_ATTR), used 61075 / 65536 bytes (93%)
║   SEGMENT  BYTES    DESCRIPTION
╠══ ICACHE   32768    reserved space for flash instruction cache
╚══ IRAM     28307    code in IRAM    
. Code in flash (default, ICACHE_FLASH_ATTR), used 249480 / 1048576 bytes (23%)
║   SEGMENT  BYTES    DESCRIPTION
╚══ IROM     249480   code in flash   
esptool.py v3.0
Serial port COM5
Connecting....
Chip is ESP8266EX
Features: WiFi
Crystal is 26MHz
MAC: 50:02:91:e0:4a:8a
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 460800
Changed.
Configuring flash size...
Auto-detected Flash size: 4MB
Compressed 285424 bytes to 208508...
Writing at 0x00000000... (7 %)
Writing at 0x00004000... (15 %)
Writing at 0x00008000... (23 %)
Writing at 0x0000c000... (30 %)
Writing at 0x00010000... (38 %)
Writing at 0x00014000... (46 %)
Writing at 0x00018000... (53 %)
Writing at 0x0001c000... (61 %)
Writing at 0x00020000... (69 %)
Writing at 0x00024000... (76 %)
Writing at 0x00028000... (84 %)
Writing at 0x0002c000... (92 %)
Writing at 0x00030000... (100 %)
Wrote 285424 bytes (208508 compressed) at 0x00000000 in 4.7 seconds (effective 490.9 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
```

Photo of the board and wiring attached.

[image]

I tried powering Sonoff with the output from the Wemos. The problem is that the moment I connect that forth (red) wire, the COM port in windows momentarly disappears leading to the Python script losing access right to that port, if that makes sense. Maybe it would have been fin in Linux, but doesn't work in Windows.

mightymos, 2025-03-18:
Ok.
I do not know if all those cables are connected to anything else.
But could you try with just the USB cable connected to the Wemos and nothing else connected?
Then power the Sonoff from the Wemos.

What is the part number on that chip with the three pins on one side and one pin on the other side near the 5V and RESET header?

pszober, 2025-03-18:
Sorry, I had to drop my son to his accordion lesson.

This is what I was trying to explain - if I power the RFbridge from Wemos, I momentarly lose connection on the COM port resulting in thouands of these lines in the python script:

```
2025-03-18 21:42:03,887 - ERROR - State 3 failed. Returning to handshake state.
2025-03-18 21:42:03,887 - ERROR - Error during check_if_ready: ClearCommError failed (PermissionError(13, 'Access is denied.', None, 5))
2025-03-18 21:42:03,887 - ERROR - State 3 failed. Returning to handshake state.
2025-03-18 21:42:03,887 - ERROR - Error during check_if_ready: ClearCommError failed (PermissionError(13, 'Access is denied.', None, 5))
2025-03-18 21:42:03,887 - ERROR - State 3 failed. Returning to handshake state.
2025-03-18 21:42:03,887 - ERROR - Error during check_if_ready: ClearCommError failed (PermissionError(13, 'Access is denied.', None, 5))
2025-03-18 21:42:03,887 - ERROR - State 3 failed. Returning to handshake state.
2025-03-18 21:42:03,888 - ERROR - Error during check_if_ready: ClearCommError failed (PermissionError(13, 'Access is denied.', None, 5))
```

The other two cables you see on the photo are just external power adapters - 12v for the Wemos and 5v for Sonoff.

mightymos, 2025-03-22:
Okay, I think the regulator should be capable of supplying 1 A so that should be plenty.
I would need to know more markings or measure the output to confirm that is the 3.3V output but it might be.

It would be simplest to start with just one USB cable supplying power to the flasher board.

There could be a brown out occurring due to too much current.
Also did you erase the ESP8285?

pszober, 2025-03-24:
Thanks for trying to help.

I did wipe the 8285 first. I repeated it just in case. Here's the output:
```
C:\Users\pszob\Downloads\esptool-win64>esptool --chip auto --port com4 erase_flash
esptool.py v4.8.1
Serial port com4
Connecting...
Failed to get PID of a device on com4, using standard reset sequence.
.
Detecting chip type... Unsupported detection protocol, switching and trying again...
Connecting...
Failed to get PID of a device on com4, using standard reset sequence.

Detecting chip type... ESP8266
Chip is ESP8285N08
Features: WiFi, Embedded Flash
Crystal is 26MHz
MAC: ec:94:cb:b3:d5:49
Uploading stub...
Running stub...
Stub running...
Erasing flash (this may take a while)...
Chip erase completed successfully in 2.5s
Hard resetting via RTS pin...

```

Next I compiled and flashed it with Arduino:
```
. Variables and constants in RAM (global, static), used 62888 / 80192 bytes (78%)
║   SEGMENT  BYTES    DESCRIPTION
╠══ DATA     1520     initialized variables
╠══ RODATA   1960     constants       
╚══ BSS      59408    zeroed variables
. Instruction RAM (IRAM_ATTR, ICACHE_RAM_ATTR), used 61075 / 65536 bytes (93%)
║   SEGMENT  BYTES    DESCRIPTION
╠══ ICACHE   32768    reserved space for flash instruction cache
╚══ IRAM     28307    code in IRAM    
. Code in flash (default, ICACHE_FLASH_ATTR), used 249480 / 1048576 bytes (23%)
║   SEGMENT  BYTES    DESCRIPTION
╚══ IROM     249480   code in flash   
esptool.py v3.0
Serial port COM5
Connecting....
Chip is ESP8266EX
Features: WiFi
Crystal is 26MHz
MAC: 50:02:91:e0:4a:8a
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 460800
Changed.
Configuring flash size...
Auto-detected Flash size: 4MB
Compressed 285424 bytes to 208507...
Writing at 0x00000000... (7 %)
Writing at 0x00004000... (15 %)
Writing at 0x00008000... (23 %)
Writing at 0x0000c000... (30 %)
Writing at 0x00010000... (38 %)
Writing at 0x00014000... (46 %)
Writing at 0x00018000... (53 %)
Writing at 0x0001c000... (61 %)
Writing at 0x00020000... (69 %)
Writing at 0x00024000... (76 %)
Writing at 0x00028000... (84 %)
Writing at 0x0002c000... (92 %)
Writing at 0x00030000... (100 %)
Wrote 285424 bytes (208507 compressed) at 0x00000000 in 4.7 seconds (effective 490.9 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...

```

Then I connected the Wemos board to the PC USB port (left all the other power sources unplugged), run the python script, connected the 3.3v pin and got this:
```
PS C:\Users\pszob\Downloads\OnbrightFlasher> python flashScript.py
Available COM ports:
1. COM1 - Communications Port (COM1)
2. COM5 - USB-SERIAL CH340 (COM5)
Select a COM port (enter the number): 2
2025-03-24 14:22:10,221 - INFO - Serial connection opened on COM5
Select a file by number:
1. blink.ihx
2. portisch_main_OB38S003_MULTIPLE_PROTOCOLS_INCLUDED.hex
Enter the number of the file you want to select: 2
2025-03-24 14:22:10,588 - INFO - You selected: C:\Users\pszob\Downloads\OnbrightFlasher\portisch_main_OB38S003_MULTIPLE_PROTOCOLS_INCLUDED.hex
2025-03-24 14:22:11,692 - INFO - handshake
2025-03-24 14:22:11,692 - INFO - State changing to handshake
2025-03-24 14:22:11,694 - INFO - cycle power to target (start with power off and then turn on)
2025-03-24 14:22:19,013 - INFO - Handshake succeeded
2025-03-24 14:22:19,133 - INFO - Status: 3
2025-03-24 14:22:19,137 - INFO - NACK on transmit of data
2025-03-24 14:22:19,139 - INFO - Chip read type FAILED
2025-03-24 14:22:19,140 - INFO - Chip type reported was: 0xD0
2025-03-24 14:22:19,145 - INFO - Can try command [signature] or [idle] then [handshake] to retry
2025-03-24 14:22:19,145 - ERROR - Something went wrong. Process ended with Can try command [signature] or [idle] then [handshake] to retry
2025-03-24 14:22:19,146 - ERROR - State 4 failed. Returning to handshake state.
2025-03-24 14:22:20,262 - INFO - handshake
2025-03-24 14:22:20,263 - INFO - State changing to handshake
2025-03-24 14:22:20,264 - INFO - cycle power to target (start with power off and then turn on)
2025-03-24 14:22:50,562 - ERROR - Timeout reached. No valid response within 30 seconds. Last response was:
2025-03-24 14:22:50,563 - ERROR - State 4 failed. Returning to handshake state.
2025-03-24 14:22:51,668 - INFO - handshake
2025-03-24 14:22:51,669 - INFO - State changing to handshake
2025-03-24 14:22:51,670 - INFO - cycle power to target (start with power off and then turn on)

```

Same happens if I try using independent power supplies. I'm out of ideas...

mightymos, 2025-03-25:
What is the baud rate set at in the Serial Monitor?

pszober, 2025-03-26:
9600 selected in Serial Monitor window. 
In Windows Device Manager, the relevant port called USB-SERIAL CH340 (COM4) is set as:
9600
8 bits
Parity none
Stop bit 1
FC none

mightymos, 2025-03-26:
Ok, the boot up baud rate is 74880 so that can cause some garbled output.
However, after that the readme says to use 115200.

It's possible your board is using a different means to convert USB to serial but you can try that 115200 baud rate first.

pszober, 2025-03-26:
We are getting somewhere! 74880 gives me this after I press the reset button on the Wemos board:
```
ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 3424, room 16 
tail 0
chksum 0x2e
load 0x3fff20b8, len 40, room 8 
tail 0
chksum 0x2b
csum 0x2b
v00045af0
~ld
rf cal sector: 1020
freq trace enable 0
rf[112] : 0��
�No�
Z���Y
�]�)l�M�@��5PT��V?I�RҺn1@�L�@�[J�Y9AN؍�(ZV?O�
#S��=A�ZA�N1A�T�P�^1%noK[�

```

I will try to wipe it now

mightymos, 2025-03-26:
Did you try 115200 baud rate?

pszober, 2025-03-26:
So here is what happens. When I reset the board, it first communicates using 74880 baund rate, but that "garbage" that follows later, is basically the board switching to 115200. So if I just leave it in 115200 it works fine, but I'm getting the same exact problem, which is reported chip type of 0xD0.

```
cycle power to target (start with power off and then turn on)
Handshake succeeded
Status: 4
Other error
Chip read type FAILED
Chip type reported was: 0xD0
Can try command [signature] or [idle] then [handshake] to retry
handshake
State changing to handshake
cycle power to target (start with power off and then turn on)
Handshake succeeded
Status: 3
NACK on transmit of data
Chip read type FAILED
Chip type reported was: 0xD0
Can try command [signature] or [idle] then [handshake] to retry

```

pszober, 2025-03-26:
It worked! Thank you so, so much.

I'm flashing it now with ESPhome. I will let you know if it all worked as intended.

:)

pszober, 2025-03-26:
Flashed with ESPhome now and it appears to be working correctly, so I'm closing the issue.

Unfortunately, it doesn't read any of my three RF remotes. One from a LED wall controller, one from an infrared heater and the last one from xmas lights. They must be all working on some different freq or protocol, but anyway that's nothing to do with this issue.

closed 2025-03-26 by pszober

mightymos, 2025-03-26:
It's probably easier to use Tasmota when attempting to view unknown codes.
People have had good success sniffing remote codes with the SNIFFING firmware.
You would issue an "rfraw AA B1 55" command in Tasmota console after placing it in Sonoff Bridge (25) mode.

If the remotes are truly at another frequency it will not work of course.
But if you're willing to try it could possibly work.

pszober, 2025-03-26:
Thank you. I'm just after desoldering the wires and putting the whole box together. I will take it apart again tomorrow and flash it with Tasmota. I find it hard to believe that all three remotes are different t frequency. I hope sniffing for non-standard codes will give me some results.

pszober, 2025-03-27:
So I reflashed it with Tasmota using the web tool. I double-checked the 'hack' soldering and cut paths and all appears to be good.

Then I downloaded RF-Bridge-EFM8BB1-20190220.hex and tried to upload it via the Tasmota web interface but I'm getting "Upload Failed Upload error code 210". I tried googling it and there is more people who had this problem with the v2.2 board but no idea how to fix it.

I tried powering it from 3.3v pins as wel las from USB, no difference.

pszober, 2025-03-27:
Oops! I followed this, so soldered two wires and cus some paths: 
https://community.home-assistant.io/t/new-sonoff-rf-bridge-board-need-flashing-help/344326/17

Let me try flashing it with OB38S003. Would you still recommend Tasmota over ESPhome? I'm asking as half my hous runs on ESPhome and not a single Tasmota, so I would prefer sticking with it.

The board right now:

[image]

[image]

mightymos, 2025-03-27:
Okay by cutting the traces you have electrically bypassed the microcontroller.
So you will have to decode radio signals directly on the ESP8265.
You can use Tasmota or ESPHome to do that.

pszober, 2025-03-27:
I have the chip flashed using the OnbrightFlasher with portisch_main_OB38S003_MULTIPLE_PROTOCOLS_INCLUDED.hex

Standard Tasmota installed on board and configured as Sonoff Bridge (25) - is that correct? Or should I use a template, as some people suggest (https://pastebin.com/raw/NByhKiXV). 

Sorry, this isn't exactly my teritory and I'm a little lost.

pszober, 2025-03-27:
It's working!!! Reading codes from my RF remotes :)

```
INFO ESPHome 2025.3.2
INFO Reading configuration /config/esphome/sonoff-rf-bridge.yaml...
INFO Starting log output from sonoff-rf.local using esphome API
INFO Successfully connected to sonoff-rf @ 192.168.1.80 in 2.749s
INFO Successful handshake with sonoff-rf @ 192.168.1.80 in 0.013s
[17:32:48][I][app:100]: ESPHome version 2025.3.2 compiled on Mar 27 2025, 17:31:42
[17:32:48][C][status_led:019]: Status LED:
[17:32:48][C][status_led:020]:   Pin: GPIO13
[17:32:48][C][wifi:600]: WiFi:
[17:32:48][C][wifi:428]:   Local MAC: EC:94:CB:B3:D5:49
[17:32:48][C][wifi:433]:   SSID: [redacted]
[17:32:48][C][wifi:436]:   IP Address: 192.168.1.80
[17:32:48][C][wifi:439]:   BSSID: [redacted]
[17:32:48][C][wifi:441]:   Hostname: 'sonoff-rf'
[17:32:48][C][wifi:443]:   Signal strength: -46 dB ▂▄▆█
[17:32:48][C][wifi:447]:   Channel: 11
[17:32:48][C][wifi:448]:   Subnet: 255.255.255.0
[17:32:48][C][wifi:449]:   Gateway: 192.168.1.1
[17:32:48][C][wifi:450]:   DNS1: 192.168.1.11
[17:32:48][C][wifi:451]:   DNS2: 1.1.1.3
[17:32:48][C][logger:177]: Logger:
[17:32:48][C][logger:178]:   Max Level: DEBUG
[17:32:48][C][logger:179]:   Initial Level: DEBUG
[17:32:48][C][logger:181]:   Log Baud Rate: 115200
[17:32:48][C][logger:182]:   Hardware UART: UART0
[17:32:48][C][remote_receiver.esp8266:060]: Remote Receiver:
[17:32:48][C][remote_receiver.esp8266:061]:   Pin: GPIO4
[17:32:48][C][remote_receiver.esp8266:066]:   Buffer Size: 1000
[17:32:48][C][remote_receiver.esp8266:067]:   Tolerance: 25%
[17:32:48][C][remote_receiver.esp8266:069]:   Filter out pulses shorter than: 50 us
[17:32:48][C][remote_receiver.esp8266:070]:   Signal is done after 10000 us of no changes
[17:32:48][C][remote_transmitter:018]: Remote Transmitter...
[17:32:48][C][remote_transmitter:019]:   Carrier Duty: 100%
[17:32:48][C][remote_transmitter:020]:   Pin: GPIO5
[17:32:48][C][web_server:285]: Web Server:
[17:32:48][C][web_server:286]:   Address: sonoff-rf.local:80
[17:32:48][C][mdns:116]: mDNS:
[17:32:48][C][mdns:117]:   Hostname: sonoff-rf
[17:32:48][C][esphome.ota:073]: Over-The-Air updates:
[17:32:48][C][esphome.ota:074]:   Address: sonoff-rf.local:8266
[17:32:48][C][esphome.ota:075]:   Version: 2
[17:32:48][C][esphome.ota:078]:   Password configured
[17:32:49][C][safe_mode:018]: Safe Mode:
[17:32:49][C][safe_mode:019]:   Boot considered successful after 60 seconds
[17:32:49][C][safe_mode:021]:   Invoke after 10 boot attempts
[17:32:49][C][safe_mode:022]:   Remain in safe mode for 300 seconds
[17:32:49][C][api:140]: API Server:
[17:32:49][C][api:141]:   Address: sonoff-rf.local:6053
[17:32:49][C][api:145]:   Using noise encryption: NO
[17:32:52][I][remote.pronto:231]: Received Pronto: data=
[17:32:52][I][remote.pronto:233]: 0000 006D 0048 0000 0008 001F 0009 001D 000A 001D 001D 000B 0009 001D 000A 001D 001C 000B 001C 000B 0009 001D 000A 001D 001C 000B 0009 001D 000A 001D 000A 001D 000A 001D 001C 000B 0009 001D 000A 001D 000A 001D 000A 001D 000A 001D 
[17:32:52][I][remote.pronto:233]: 000A 001D 000A 001D 000A 0152 001C 000B 0009 001D 000A 001D 001C 000B 0009 001D 000A 001D 001D 000B 001C 000B 0009 001D 000A 001D 001C 000B 0009 001D 000A 001D 000A 001D 000A 001D 001C 000B 0009 001D 000A 001D 000A 001D 000A 001D 
[17:32:52][I][remote.pronto:233]: 000A 001D 000A 001D 000A 001D 000A 0152 001C 000B 0009 001D 000A 001D 001C 000B 0009 001D 000A 001D 001C 000B 001C 000B 0009 001D 000A 001D 001C 000B 0009 001D 000A 001D 000A 001D 0009 001D 001D 000B 0009 001D 000A 001D 000A 001D 
[17:32:52][I][remote.pronto:233]: 000A 001D 000A 001D 000A 001D 000A 001D 000A 0181 
[17:32:52][W][component:237]: Component remote_receiver took a long time for an operation (77 ms).
[17:32:52][W][component:238]: Components should block for at most 30 ms.
[17:32:52][I][remote.pronto:231]: Received Pronto: data=
[17:32:52][I][remote.pronto:233]: 0000 006D 0001 0000 0152 0181 
[17:32:54][I][remote.canalsatld:099]: Received CanalSatLD: device=0x0C, address=0x39, command=0x3C, repeat=0x1
[17:32:54][I][remote.pronto:231]: Received Pronto: data=
[17:32:54][I][remote.pronto:233]: 0000 006D 0090 0000 0009 001E 0009 001D 000A 001D 001D 000B 0009 001D 000A 001D 001C 000B 001C 000B 0009 001D 000A 001D 001C 000B 0009 001D 000A 001D 001C 000B 001C 000B 001C 000B 0009 001D 000A 001D 000A 001D 000A 001D 000A 001D 
[17:32:54][I][remote.pronto:233]: 000A 001D 000A 001D 000A 0152 001C 000B 0009 001D 000A 001D 001D 000B 0009 001D 000A 001D 001D 000B 001C 000B 0009 001D 000A 001D 001D 000B 0009 001D 000A 001D 001C 000B 001C 000B 001C 000B 0009 001D 0009 001D 0009 001E 0009 001D 
[17:32:54][I][remote.pronto:233]: 000A 001D 000A 001D 000A 001D 000A 0153 001C 000B 0009 001D 000A 001D 001D 000B 0009 001D 000A 001D 001C 000B 001C 000B 0009 001D 000A 001D 001D 000B 0009 001D 000A 001D 001D 000B 001C 000B 001C 000B 0009 001D 000A 001D 000A 001D 
[17:32:54][I][remote.pronto:233]: 000A 001D 000A 001D 000A 001D 000A 001D 000A 0152 001C 000B 0009 001D 000A 001D 001D 000B 0009 001E 0009 001D 001C 000B 001C 000B 0009 001D 000A 001D 001C 000B 0009 001D 000A 001D 001D 000B 001C 000B 001C 000B 0009 001D 000A 001D 
[17:32:54][I][remote.pronto:233]: 000A 001D 000A 001D 000A 001D 000A 001D 000A 001D 000A 0152 001C 000B 0009 001D 000A 001D 001C 000B 0009 001D 000A 001D 001D 000B 001C 000B 0009 001D 000A 001D 001C 000B 0009 001D 000A 001D 001D 000B 001C 000B 001C 000B 0009 001D 
[... cut ...]
[17:32:56][I][remote.pronto:233]: 0000 006D 0001 0000 0152 0181 

```

I followed mostly [this](https://community.home-assistant.io/t/new-sonoff-rf-bridge-board-need-flashing-help/344326) guide, with few things borrowed from your guide.

Next I will need to figure out how to actually now use it, so that the codes for different buttons can be learned and used in automation, but for that I imagine there will be other tutorials and posts. Any links appreciated. I will stick with ESPhome.

Once again, thank you so much for all your assistance (and patience).

mightymos, 2025-03-27:
Great, good luck.