mightymos/OnbrightFlasher #14: Problems with falshing

sstefanowski, 2025-04-01:
Hello,

First of all thank you for preparing this software. 
I wanted to try flashing my Sonoff RF v 2.2. with D1 R2 and got stuck on erasing MCU chip

I'm using Windows 11. 
When I do the procedure via Serial Monitor, I get an error:
note (chip type is 0xD0 not 0xA
```
State changing to handshake
Cycle power to target (start with power off and then turn on)

Handshake succeeded
Status: 4
Other error
Chip read type FAILED
Chip type reported was: 0xD0
Can try command [signature] or [idle] then [handshake] to retry
erase
Erasing chip...
Status: 2
NACK on transmit of address
Chip erase FAILED
```

When I try flashscript.py I got the following errors:
```
025-04-01 01:06:10,318 - INFO - You selected: F:\GIT\OnbrightFlasher\portisch_main_OB38S003_MULTIPLE_PROTOCOLS_INCLUDED.hex
2025-04-01 01:06:11,423 - INFO - handshake
2025-04-01 01:06:11,426 - INFO - State changing to handshake
2025-04-01 01:06:11,432 - INFO - cycle power to target (start with power off and then turn on)
2025-04-01 01:06:20,581 - INFO - Handshake succeeded
2025-04-01 01:06:20,700 - INFO - Status: 2
2025-04-01 01:06:20,701 - ERROR - Something went wrong. Process ended with Status: 2
2025-04-01 01:06:20,705 - ERROR - State 4 failed. Returning to handshake state.
2025-04-01 01:06:20,709 - INFO - NACK on transmit of address
2025-04-01 01:06:20,712 - INFO - Chip read type FAILED
2025-04-01 01:06:20,715 - INFO - Chip type reported was: 0xD0
2025-04-01 01:06:20,723 - INFO - Can try command [signature] or [idle] then [handshake] to retry
2025-04-01 01:06:20,724 - ERROR - Something went wrong. Process ended with Can try command [signature] or [idle] then [handshake] to retry
2025-04-01 01:06:20,726 - ERROR - State 3 failed. Returning to handshake state.
2025-04-01 01:06:21,843 - INFO - handshake
2025-04-01 01:06:21,847 - INFO - State changing to handshake
2025-04-01 01:06:21,853 - INFO - cycle power to target (start with power off and then turn on)
2025-04-01 01:06:22,994 - ERROR - Error during check_if_ready: GetOverlappedResult failed (PermissionError(13, 'Access is denied.', None, 5))
2025-04-01 01:06:22,995 - ERROR - State 4 failed. Returning to handshake state.
2025-04-01 01:06:22,998 - ERROR - Error during check_if_ready: ClearCommError failed (PermissionError(13, 'Access is denied.', None, 5))
2025-04-01 01:06:23,000 - ERROR - State 3 failed. Returning to handshake state.
```

I'm pretty sure I connected the wires correctly.
Any ideas?

mightymos, 2025-04-01:
For whatever reason it seems like people have been needing to erase manually.
Just use the Serial Monitor in the arduino software at 115200 baud and following handshake try [erase].

It is strange you are getting a permission error.
It is possible you need to run something as an administrator.

sstefanowski, 2025-04-01:
I tried running flashscript.py as Admninistrator and I failed.

It would be strange that Arduino IDE was able to upload your software to my D1 R2, but it is unable to do flashing, wouldn't it?
Anyway. I spent like 4 hours failing, 
I'll try today in 10 hours again.... with manual procedure starting from beginning, erasing ESP8256 and other retrying erase...
I'll also post pictures of my boards and wiring. If it does not help I;ll try to get new NodeMCU for retrying.

DO you think it may be hardware problem?

sstefanowski, 2025-04-01:
BTW. Also, Yesterday, I found one of my cables connecting D1 to Sonoff was loose (on Sonoff's side), so I replaced all 4 cables connecting devices without much luck.
Do you think I accidentally might "cook" it if it accidentally disconnected, e.g., during erase? Do you have any recommendations for checking hardware?

Note: I'm pretty sure I always used the 3.3V pin on the D1 side and never was in risk of connecting to 5V.

mightymos, 2025-04-01:
It's difficult to say definitely what could be happening without measuring something (e.g. with an oscilloscope).
What board are you using as a flasher?

sstefanowski, 2025-04-01:
I got a little progress. I manged to erase MCU but flashing still fails? Hex file problem?

What I I changed: 
- connected the power of Sonof via MniUSB instead of 3.3V pin
- changed SDA and SCL pins from to the ones on the side of D1 board (previously it was connected to the SDA and SCL pins in the middle). Was it a problem?

Now, I can errase MCU successfully but flashing `portisch_main_OB38S003_MULTIPLE_PROTOCOLS_INCLUDED.hex` fails with this log:
```
2025-04-01 18:42:36,378 - INFO - Serial connection opened on COM3
2025-04-01 18:42:38,643 - INFO - You selected: F:\GIT\OnbrightFlasher\portisch_main_OB38S003_MULTIPLE_PROTOCOLS_INCLUDED.hex
2025-04-01 18:42:39,757 - INFO - handshake
2025-04-01 18:42:39,759 - INFO - State changing to handshake
2025-04-01 18:42:39,762 - INFO - cycle power to target (start with power off and then turn on)
2025-04-01 18:42:43,520 - INFO - Handshake succeeded
2025-04-01 18:42:43,639 - INFO - Status: 0
2025-04-01 18:42:43,640 - INFO - Success
2025-04-01 18:42:43,642 - INFO - Chip read: 0xA
2025-04-01 18:42:43,643 - INFO - Connected...
2025-04-01 18:42:43,644 - INFO - Returning to idle state...
2025-04-01 18:42:43,746 - INFO - erase
2025-04-01 18:42:44,248 - INFO - Erasing chip...
2025-04-01 18:42:44,250 - INFO - Status: 0
2025-04-01 18:42:44,253 - INFO - Success
2025-04-01 18:42:44,256 - INFO - Chip erase successful
2025-04-01 18:42:44,357 - INFO - 
2025-04-01 18:42:44,859 - INFO - setfuse 18 249
2025-04-01 18:42:45,361 - INFO - Set configuration byte...
2025-04-01 18:42:45,362 - INFO - Status: 0
2025-04-01 18:42:45,362 - INFO - Success
2025-04-01 18:42:45,363 - INFO - Wrote configuration byte
2025-04-01 18:42:45,368 - INFO - :0400000002009E322A
2025-04-01 18:42:45,373 - INFO - Write successful
2025-04-01 18:42:45,374 - INFO - Wrote 4 bytes
2025-04-01 18:42:45,378 - INFO - :03000B00021817C1
2025-04-01 18:42:45,382 - INFO - Write successful
2025-04-01 18:42:45,383 - INFO - Wrote 3 bytes
2025-04-01 18:42:45,388 - INFO - :0100130032BA
2025-04-01 18:42:45,391 - INFO - Write successful
2025-04-01 18:42:45,391 - INFO - Wrote 1 bytes
2025-04-01 18:42:45,395 - INFO - :03001B000218596F
2025-04-01 18:42:45,400 - INFO - Write successful
2025-04-01 18:42:45,401 - INFO - Wrote 3 bytes
2025-04-01 18:42:45,405 - INFO - :030023000218E9D7
2025-04-01 18:42:45,409 - INFO - Write successful
2025-04-01 18:42:45,410 - INFO - Wrote 3 bytes
2025-04-01 18:42:45,414 - INFO - :03002B0002189B1D
2025-04-01 18:42:45,418 - INFO - Write successful
2025-04-01 18:42:45,419 - INFO - Wrote 3 bytes
2025-04-01 18:42:45,425 - INFO - :100030000000E2FBEAF2802C0000E0FBEAF0802402
2025-04-01 18:42:45,442 - INFO - Write successful
2025-04-01 18:42:45,443 - INFO - Wrote 16 bytes
2025-04-01 18:42:45,449 - INFO - :10004000E6B50202EBF62200E2B50202EBF2220074
2025-04-01 18:42:45,466 - INFO - Write successful
2025-04-01 18:42:45,467 - INFO - Wrote 16 bytes
2025-04-01 18:42:45,473 - INFO - :10005000E0B50202EBF02230F6E0A88220F5D3EA08
2025-04-01 18:42:45,489 - INFO - Write successful
2025-04-01 18:42:45,490 - INFO - Wrote 16 bytes
2025-04-01 18:42:45,497 - INFO - :10006000C6F582228B822230F6E6A88220F5D9805E
2025-04-01 18:42:45,514 - INFO - Write successful
2025-04-01 18:42:45,516 - INFO - Wrote 16 bytes
2025-04-01 18:42:45,522 - INFO - :01007000CFC0
2025-04-01 18:42:45,523 - INFO - Write successful
2025-04-01 18:42:45,525 - INFO - Wrote 1 bytes
2025-04-01 18:42:45,531 - INFO - :0600F700750A0075090006
2025-04-01 18:42:45,537 - INFO - Write successful
2025-04-01 18:42:45,539 - INFO - Wrote 6 bytes
2025-04-01 18:42:45,543 - INFO - :0301000002007189
2025-04-01 18:42:45,547 - INFO - Write successful
2025-04-01 18:42:45,548 - INFO - Wrote 3 bytes
2025-04-01 18:42:45,552 - INFO - :030071000204A2E4
2025-04-01 18:42:45,556 - INFO - Write successful
2025-04-01 18:42:45,557 - INFO - Wrote 3 bytes
2025-04-01 18:42:45,564 - INFO - :10010300AF829000C2E0FE24FB500122EE2E2E901F
2025-04-01 18:42:45,569 - INFO - Status: 2
2025-04-01 18:42:45,570 - ERROR - Error: Unexpected response or mismatch with sent line.
2025-04-01 18:42:45,570 - ERROR - Sent line representation: ':10010300AF829000C2E0FE24FB500122EE2E2E901F'
2025-04-01 18:42:45,571 - ERROR - Response representation: 'Status: 2'
2025-04-01 18:42:45,571 - ERROR - State 8 failed. Returning to handshake state.
2025-04-01 18:42:45,572 - INFO - NACK on transmit of address
2025-04-01 18:42:45,580 - INFO - Write FAILED
2025-04-01 18:42:45,581 - INFO - Errors: 1
2025-04-01 18:42:45,584 - INFO - [can try sending hex line again]
2025-04-01 18:42:46,694 - INFO - handshake
2025-04-01 18:42:46,695 - INFO - State changing to handshake
2025-04-01 18:42:46,696 - INFO - cycle power to target (start with power off and then turn on)
2025-04-01 18:43:17,075 - ERROR - Timeout reached. No valid response within 30 seconds. Last response was: 
2025-04-01 18:43:17,076 - ERROR - State 4 failed. Returning to handshake state.
```

Any ideas how to push it forward???

-----

My Board is D1 clone with ESP8266EX chip, In ARDUINO IDE I selected "LOLIN (WEBMOS) D1 R2 & mini"

[image]
[image]

[image]

mightymos, 2025-04-01:
I believe that the SDA and SCL pins are all connected together on that board:
https://forum.arduino.cc/t/wemos-d1-r2-duplicate-pins/1160953

But to know with certainty you would have to use a multimeter or visually inspect the board traces.

It is possible you have an intermittent connection because the jumper wires are not connecting to a soldered header.
There also needs to be a ground connected between both boards.
The USB connections are probably providing the same ground if they are both plugged into the same source.

However, it is probably best to connect grounds between boards with a jumper wire and power the target board with 3.3V output from the flasher board.

mightymos, 2025-04-01:
> Wooow, success!
> 
> I added GDN wire between two boards and was successful, while providing power to Sonnof still from USB. Previously both boards were powered from two USB ports of the same laptop... but it was not good.
> 
> BTW I tries power 3.3V + GND from D1 to Sonoff but probably was unable to do it quickly enough. Using power from additional USB Mini make it quicker for me.
> 
> Looks like connecting with just SCL and SDA is not good idea idea and additional GND gives better synchronization!

Fantastic, good news.

closed 2025-05-05 by mightymos