mightymos/OnbrightFlasher #15: Compiling for ATMega 2560 fails "Not Enough Memory"

GJChristman, 2025-05-05:
Greetings,
As per the title, I cannot seem to compile the OnbrightFlasher for an ATMega 2560 board that I purchased.

I have coded for microcontrollers (Nordic NRF, Pics etc.) but I'm not familiar at all with the Arduino environment.

I installed the Arduino IDE and can connect to the board, but the compiled image uses about 32K of RAM, which is way more than the 8K that the ATMega has.  

Is there something I'm missing?  The ReadMe states that it has been tested with this board.

Any pointers would be appreciated.

-Glenn

mightymos, 2025-05-05:
That does not sound quite right, but it has been awhile so maybe my memory is mistaken.
I did find my comment from the past:
https://github.com/mightymos/OnbrightFlasher/issues/10#issuecomment-2510614350

You could try following the advice from that comment and see if that reduces RAM footprint?

GJChristman, 2025-05-05:
Looks like that did it - I changed the FILE_ARRAY_MAX size  in onbrightFlasher.h  to 500 from 32727 and it verifies.

Many thanks for the help!

mightymos, 2025-05-05:
Great news, closing as resolved for now.

closed 2025-05-05 by mightymos