mightymos/OnbrightFlasher #16: Unable to erase MCU

jeffemandel, 2025-06-06:
I'm following the instructions in the "OnbrightFlasher/flashing by example"

I have a brand new RF BridgeR2 V2.2, and I successfully erased the ESP8256 on the RF bridge.

I have my brand new WeMos D1 mini, and the latest OnbrightFlasher compiled and installed. I soldered headers to the J3 pins, and have SDA/SDL connected from the D1 to the MCU. I'm powering the MCU from my USB UART. 
I connect to the serial monitor, issue handshake, and power up the MCU. I get:
````
handshake
State changing to handshake
cycle power to target (start with power off and then turn on)

Handshake succeeded
Status: 2
NACK on transmit of address
Chip read type FAILED
Chip type reported was: 0xD0
Can try command [signature] or [idle] then [handshake] to retry
erase
Erasing chip...
Status: 2
NACK on transmit of address
Chip erase FAILED

signature
Read chip type...
Status: 2
NACK on transmit of address
Chip read type FAILED
Chip type reported was: 0x7C
````
Suggestions?

mightymos, 2025-06-09:
Double check that SDA/SCL are not swapped, nor offset by one pin etc.
I am not sure, but could you try powering the Sonoff a different way?

My intent would be that ground is always connected and you just connect/disconnect the supply voltage.
I no longer have this hardware and so cannot test it personally.

jeffemandel, 2025-06-09:
> mightymos

￼

Here is my setup. I switched to powering from the USB UART because the powering off the Wemos caused it to reset. 

I tried installing Tasmoto with tasmoto-sensors.bin, switched to Generic, and set the pins as suggested in https://github.com/mightymos/RF-Bridge-OB38S003?tab=readme-ov-file#tasmotaesphomeespurna

Using this configuration, I was able to handshake and erase using OnbrightFlasher.ino 
I then tried flashscript.py:

Select a COM port (enter the number): 4
2025-06-09 16:32:50,287 - INFO - Serial connection opened on /dev/cu.usbserial-2310
Select a file by number:
1. blink.ihx
2. passthrough_main_OB38S003.hex
Enter the number of the file you want to select: 2
2025-06-09 16:32:55,425 - INFO - You selected: /Users/jeffemandel/Documents/Arduino/OnbrightFlasher/passthrough_main_OB38S003.hex
2025-06-09 16:32:56,534 - INFO - handshake
2025-06-09 16:32:56,535 - INFO - State changing to handshake
2025-06-09 16:32:56,536 - INFO - cycle power to target (start with power off and then turn on)
2025-06-09 16:33:07,308 - INFO - Handshake succeeded
2025-06-09 16:33:07,427 - INFO - Status: 2
2025-06-09 16:33:07,428 - ERROR - Something went wrong. Process ended with Status: 2
2025-06-09 16:33:07,428 - ERROR - State 4 failed. Returning to handshake state.
2025-06-09 16:33:07,430 - INFO - NACK on transmit of address
2025-06-09 16:33:07,430 - INFO - Chip read type FAILED
2025-06-09 16:33:07,433 - INFO - Chip type reported was: 0xD0
2025-06-09 16:33:07,440 - INFO - Can try command [signature] or [idle] then [handshake] to retry
2025-06-09 16:33:07,440 - ERROR - Something went wrong. Process ended with Can try command [signature] or [idle] then [handshake] to retry
2025-06-09 16:33:07,440 - ERROR - State 3 failed. Returning to handshake state.

I’ve tried this many times without success.

Suggestions?

axa88, 2025-06-09:
Just tried this myself using the "manual" method. I get this error when attempting to handshake

```
E (929409) i2c.master: probe device timeout. Please check if xfer_timeout_ms and pull-ups are correctly set up
E (929419) i2c.master: I2C hardware NACK detected
E (929419) i2c.master: I2C transaction unexpected nack detected
E (929419) i2c.master: s_i2c_synchronous_transaction(924): I2C transaction failed
E (929425) i2c.master: i2c_master_receive(1240): I2C transaction failed
Handshake succeeded
E (929561) i2c.master: I2C hardware NACK detected
E (929561) i2c.master: I2C transaction unexpected nack detected
E (929561) i2c.master: s_i2c_synchronous_transaction(924): I2C transaction failed
E (929567) i2c.master: i2c_master_multi_buffer_transmit(1186): I2C transaction failed
E (929575) i2c.master: I2C hardware NACK detected
E (929579) i2c.master: I2C transaction unexpected nack detected
E (929585) i2c.master: s_i2c_synchronous_transaction(924): I2C transaction failed
E (929592) i2c.master: i2c_master_receive(1240): I2C transaction failed
Status: 4
Other error
Chip read type FAILED
Chip type reported was: 0x3F
Can try command [signature] or [idle] then [handshake] to retry
```

I then try to erase:
```
erase
Erasing chip...
E (695622) i2c.master: I2C hardware NACK detected
E (695622) i2c.master: I2C transaction unexpected nack detected
E (695623) i2c.master: s_i2c_synchronous_transaction(924): I2C transaction failed
E (695630) i2c.master: i2c_master_multi_buffer_transmit(1186): I2C transaction failed
Status: 4
Other error
Chip erase FAILED
```

And that doesn't work..

axa88, 2025-06-09:
just tried with py script get basically the same error.
tried with different power sources...
doubtful wiring is a problem , everything is soldered, been soldering forever...

attempting on wroom32d and wrover-e have the same result.

@mightymos 
if you have any direction on monitoring the bus i might be able to hookup pulseview and provide some data points...

axa88, 2025-06-10:
f me... well thats not my problem... ill now have to open a new issue

axa88, 2025-06-10:
> What version of the Espressif library are you all using?

No idea, ive never used arduino for anything until i needed it for this so i guess whatever the latest is...
I checked the Board Manager it says 3.2.0, is that what you asked?