mightymos/OnbrightFlasher #17: Failure to flash , status 4 .sr/.pvs attached

axa88, 2025-06-10:
Im unable to flash Sonoff RF BridgeR2

Attempt with both the py script and "manual" method results are the same, `status 4`

```
E (929409) i2c.master: probe device timeout. Please check if xfer_timeout_ms and pull-ups are correctly set up
E (929419) i2c.master: I2C hardware NACK detected
E (929419) i2c.master: I2C transaction unexpected nack detected
E (929419) i2c.master: s_i2c_synchronous_transaction(924): I2C transaction failed
E (929425) i2c.master: i2c_master_receive(1240): I2C transaction failed
Handshake succeeded
E (929561) i2c.master: I2C hardware NACK detected
E (929561) i2c.master: I2C transaction unexpected nack detected
E (929561) i2c.master: s_i2c_synchronous_transaction(924): I2C transaction failed
E (929567) i2c.master: i2c_master_multi_buffer_transmit(1186): I2C transaction failed
E (929575) i2c.master: I2C hardware NACK detected
E (929579) i2c.master: I2C transaction unexpected nack detected
E (929585) i2c.master: s_i2c_synchronous_transaction(924): I2C transaction failed
E (929592) i2c.master: i2c_master_receive(1240): I2C transaction failed
Status: 4
Other error
Chip read type FAILED
Chip type reported was: 0x3F
Can try command [signature] or [idle] then [handshake] to retry
```

Attempting the erase command fails as well

```
erase
Erasing chip...
E (695622) i2c.master: I2C hardware NACK detected
E (695622) i2c.master: I2C transaction unexpected nack detected
E (695623) i2c.master: s_i2c_synchronous_transaction(924): I2C transaction failed
E (695630) i2c.master: i2c_master_multi_buffer_transmit(1186): I2C transaction failed
Status: 4
Other error
Chip erase FAILED
```

A number of methods were applied, power from flash board regulator, power from USB bus.
Made sure esp was 'erased' with minimal esphome program.

Bus seems to write 0x7C, 0x7C, 0x7C, 0x7E, then gives up.
Attached is the sigrock session:

[Status 4.zip](https://github.com/user-attachments/files/20679729/Status.4.zip)

@mightymos think you could take a look at the bus cap and see if there is anything we can do about it? thanks

mightymos, 2025-06-14:
What board are you using as a flasher?

axa88, 2025-06-18:
> What board are you using as a flasher?

I've now tried both a wroom d and wrover e, this capture was from the latter.

Any advice?

mightymos, 2025-06-18:
I believe those WROOM use ESP32 chips, let me know if that is correct.
If so there is apparently an I2C bug with version 3.2.0 in Arduino.

So you have to use version 3.1.2 or maybe 3.1.3:
https://github.com/espressif/arduino-esp32/issues/11228#issuecomment-2835070623

mightymos, 2025-06-20:
Great, glad it finally worked out.