mightymos/OnbrightFlasher #19: Documentation for configuration bytes?

mjadczak, 2025-08-03:
The datasheet in https://github.com/mightymos/RF-Bridge-OB38S003 is invaluable for hacking on the code for this chip, but it only describes what is possible to set via the configuration bytes via flashing, and not _how_ to achieve those settings.

It seems like this flasher currently issues `setfuse 18 249` while flashing, which seems to indicate that the author had access to some documentation around these config bytes / fuses? Is this documentation that is able to be shared somewhere?

mightymos, 2025-08-03:
The bytes were reverse engineered and documented here:
https://github.com/mightymos/OnbrightFlasher/wiki

I eventually wanted to support read-write-modify type operations but never got around to it.

I actually do not have this hardware anymore and so cannot do anymore work on it.
However hopefully the documentation is a good start.

mjadczak, 2025-08-04:
Ah, I hadn’t spotted the wiki - thanks for this and for all your work on this project in general!

closed 2025-08-04 by mjadczak